DomCR/ACadSharp #3: Slow Read

AhmedZero, 2021-12-07:
Hello,
Good Work on that project.
there is a problem with its speed.
with my best regards.

DomCR, 2021-12-18:
Hi @AhmedZero,
the reader is in an early stage, I've updated the readme page with useful information about the current implementation.
Regarding the reader speed, right now is not a priority, the current focus is to finish the reader to be fully compatible with all the entities.

Thanks for the feedback, I'll keep in mind your comment.

DJGosnell, 2022-02-18:
Doing some research and I've found that this has to do with the CRC8StreamHandler class.  The Read method calculates a CRC on each byte read individually and this adds up.  The Seed property in this class was read and written to over 299 million times for this simple water fountain DWG attached below.  See the performance checks below.  Once I disabled the CRC checks, the reading speed dramatically increased.  See the comparison below.

I looked around and I did not see the `CRC8StreamHandler.Seed` property being used.  I'm presuming this will be used for future verification of file integrity?

**CRC Calculation Enabled**
[image]

**CRC Calculation Disabled**
[image]

[Water Fountain DWG edfp217c_t.zip](https://github.com/DomCR/ACadSharp/files/8099869/edfp217c_t.zip)

DomCR, 2022-02-19:
@DJGosnell thanks a lot for this investigation!

Answering you question, the CRC8StreamHandler is meant to check the file integraty as is documented in the OpenDesign specification for dwg, but it seems that the reader can skip this for all the compatible versions.

To solve this issue and speed up the reader I've set up a flag in case the user wants to run the CRC checks for an specific file, the fix is in this branch : [ISSUE3_Slow-Read_Fix](https://github.com/DomCR/ACadSharp/tree/ISSUE3_Slow-Read_Fix)

Once this is merge I'll close this issue.

Thank you all!

closed 2022-02-19 by DomCR