DomCR/ACadSharp #10: How to get cad block location

chuongmep, 2022-03-05:
Hi @DomCR 
Many thanks about this project, i have a problem when try get a location of cad block, how we can do that ? 
Is this a development project in progress with it?
[image]

DomCR, 2022-03-06:
Hi @chuongmep 

I'm guessing that the selection that you are showing in the image is an insert entity in the Model space in Autocad, first you need the ```BlockRecord``` called *Model_Space in the ```Document.BlockRecords``` which contains all the entities in the model space of Autocad, there you can look for the ```Insert``` entity that relates to this Block and check the property ```Insert.InsertPoint```

I hope this helps, thanks for the support.

DomCR, 2022-03-06:
I've also created a release fixing the link between the ```Entities.Insert``` and the ```BlockRecord```
[release](https://github.com/DomCR/ACadSharp/releases/tag/v1.1.1-alpha)

chuongmep, 2022-03-06:
@DomCR you have any sample for this , i have try but it return null object 
``` cs
 string file = @"C:\Users\Chuong.Ho\Desktop\Drawing1.dwg";
            using (DwgReader reader = new DwgReader(file))
            {
                CadDocument doc = reader.Read();
                foreach (BlockRecord blockRecord in doc.BlockRecords)
                {
                    if (blockRecord.Name == "xxx")
                    {
                        Console.WriteLine(blockRecord.Name);
                        CadObject cadObject = blockRecord.BlockEntity;
                        Insert insert = cadObject as Insert;
                        Console.WriteLine(insert.ObjectName);

                    }
                }
            }
            //Console.WriteLine($"file read : {file}");
            Console.ReadLine();
```

DomCR, 2022-03-06:
Try this:

``` cs
static IEnumerable<Insert> getInsertEntities(string file, string blockname)
	{
		CadDocument doc = DwgReader.Read(file);

		// Get the model space where all the drawing entities are
		BlockRecord modelSpace = doc.BlockRecords["*Model_Space"];

		// Get the insert instance that is using the block that you are looking for
		return modelSpace.Entities.OfType<Insert>().Where(e => e.Block.Name == blockname);
	}
```

This method gets all the ```Insert``` instances in the drawing using an specific ```Block```.
For more information about how the document is organized check this [link](https://help.autodesk.com/view/OARX/2021/ENU/?guid=GUID-A809CD71-4655-44E2-B674-1FE200B9FE30)

chuongmep, 2022-03-06:
@DomCR it working for me, thanks so much !

closed 2022-03-06 by chuongmep