DomCR/ACadSharp #21: Read Geometry data

alphonse-lin, 2022-05-07:
Thank you so much for your code, it did write pretty well.

I am curious about, where can I find geometry data when I read a CAD file.

Right now I just got **CadDocument doc**, but I can not find the geometry data. Is it anything I did wrong?

DomCR, 2022-05-07:
Hi @alphonse-lin 

I'm guessing that you want to access to the entities in the model space of the document, I've added an example to show how can you access to all of them, please check the branch [Code-examples](https://github.com/DomCR/ACadSharp/tree/Code-examples), you can find a simple [example](https://github.com/DomCR/ACadSharp/blob/Code-examples/ACadSharp.Examples/DocumentExplorationExamples.cs) check the method `GetAllEntitiesInModel` this example returns all the entities drawn in the model; lines, points, inserts...

I hope this helps, if you have any other doubt please feel free to ask.

alphonse-lin, 2022-05-07:
Thank you so much, it is quite clear, also it worked for my dwg file.

closed 2022-05-07 by alphonse-lin