DomCR/ACadSharp #35: DWG Format Support Endeavors

DJGosnell, 2022-06-28:
I would like to thank you for your work on this project.  Developing a DXF/DWG interface is a whole lot of work as the format has evolved so much over the years and has many dark corners and black boxes.  I see that this project is primarily a DXF (binary & text) project but I see that you are also working on a DWG writing branch.

The DWG format is proprietary and this has stifled innovation in this area for so many years.  LibreCAD is GPL so that code is essentially untouchable for anything commercial, AutoDesk licenses their DWG software out on a subscription and the ODA has its own pitfalls.
If an open source DWG file editor was available, I feel that this would be something that could be really revolutionary for people consuming DWG files.

What would you like in support to further pursue this area of DWG reading/writing development?

DomCR, 2022-07-02:
Hi @DJGosnell,
there are multiple ways to support this project and help it's development like:
- report bugs 
- propose features that you would like

DomCR, 2022-07-02:
Hi @DJGosnell,
there are multiple ways to support this project and help it's development like:
- Report bugs or problems
- Propose features 
- Identify missing information not catch by the readers
- Create PRs with implementations

Also the you can sponsor this project with donation.
My current focus for is to finish the `DxfWriter` so the library will allow to fully read a dwg/dxf and save a dxf file, right now the `DwgReader` and `DxfReader` can read almost all the information stored in the files.

Thanks a lot for the support.

DJGosnell, 2022-07-22:
@DomCR Thank you for the reply.  I think I will focus on supporting your developments on the specifications as I don't have as much time to delve into the specs.  The main area that I'll want to focus on first is performance as I noticed that reading is quite memory hungry at times.  
Do you expect to still be making major changes to the structure of the project or should I be okay to start investigating the paths and potential performance enhancements?

I have a few questions regarding support.  Is it your intention to continue support for the .NET Framework in addition to the newer .NET Core?  I ask this because this will really hamper some of the ability to increase performance in some areas moving forward (Span<T>, stackalloc, etc).

DomCR, 2022-07-25:
Perfect, any help will be welcome! 

As for your questions:
- **Project structure**: I'm always open to changes and improvements, a project restructure will mean a v2 for the library, is usually a ton of work so it has to be worth the effort.
- **Performance enhancements**: if you work in this area I think it will be very productive, my main focus right now are the missing features and the error fix, the library is not complete so this is a priority for me.
- **.NET Framework support**: the library is simple enough to support multiple environments and I'll will try to support multiple environments if is possible, but I'm aware that .NET Core is newer and seems to have more support so the focus may change to this one.

socketing, 2022-07-28:
> 完美，任何帮助都将受到欢迎！
> 
> 至于你的问题：
> 
> * **项目结构**：我总是对更改和改进持开放态度，项目重组将意味着库的v2，通常是大量的工作，所以它必须是值得的。
> * **性能增强**：如果你在这个领域工作，我认为它会非常有成效，我现在的主要关注点是缺少的功能和错误修复，库不完整，所以这对我来说是一个优先事项。
> * **.NET Framework支持**：该库足够简单，可以支持多个环境，如果可能的话，我将尝试支持多个环境，但我知道.NET Core更新，似乎有更多的支持，因此重点可能会转移到这个环境。

Can this library read DWG files now? I see that you are still maintaining it, so I want to know whether it has been completed?

DomCR, 2022-07-29:
Hi @socketing 

The library can read almost all the information in a DWG, please check the readme file for more information about it, you will find the section `Dwg entities not implemented` which indicates some of the elements that are not supported at the moment, if you need any of them specifically you can open an issue with a request.

Here are some examples about how to use the library : https://github.com/DomCR/ACadSharp/tree/master/ACadSharp.Examples

socketing, 2022-07-31:
@DomCR   I can't run this c# example project. It directly reports an error. No dependencies were found for the project. You mean that it can read all the information of DWG, can it be displayed? It seems to me that the code can only read information instead of displaying graphics. I want to import the DWG file and display the drawing in the window.

socketing, 2022-07-31:
@DomCR  If you can, please tell me your email and I will send you an error prompt.

closed 2022-08-01 by DJGosnell