DomCR/ACadSharp #41: There are some problems with the  read3bits  function

CraneXHu, 2022-07-21:
In the opendesign specification,  3B is a sequence of 1 to 3 bits. Keep reading bits until a zero bit is encountered or until the 3rd bit is read, whatever comes first.

The implementation of the read3bits function in libredwg project is blow.
```
/** Read 1-3 bits
 *  Keep reading bits until a zero bit is encountered, => 0,2,6,7.
 *  0: 0, 10: 2, 110: 6, 111: 7. 100 for 4 or 101 for 5 is invalid.
 */
BITCODE_3B
bit_read_3B (Bit_Chain *dat)
{
  BITCODE_3B result = bit_read_B (dat);
  if (result)
    {
      BITCODE_3B next = bit_read_B (dat);
      if (next)
        {
          next = bit_read_B (dat);
          return next ? 7 : 6;
        }
      else
        {
          return 2;
        }
    }
  else
    {
      return 0;
    }
}
```

DomCR, 2022-07-22:
Hi @CraneXHu,

The method `read3Bits` is only used to read a `BitLongLong`, which is only used in the dwg files with version higher than `AC1024`, the thing is that in the `BitLongLong` all 3 bits are needed to read the length of the number, so the 3bits are documented to be read as you say if the file requires it, but this method is not used at all to read the file. Notice that in my code this method is `private`.

I've looked in `librecad` and they have a similar issue with that :
```c++
bit_read_3BLL (Bit_Chain *dat); /*unused but as documented*/
```
I'm guessing that this method is there only for the documentation but is not used at all, if they where to use it as the way is written it would fail when reading the `BitLongLong` length.

CraneXHu, 2022-07-23:
@DomCR 
I see. Thank you for your reply.

closed 2022-07-24 by DomCR