DomCR/ACadSharp #42: Ellipse - ReadDWG - MajorAxis is always 0,0,0

mattbrownfaro, 2022-07-22:
[ellipse.zip](https://github.com/DomCR/ACadSharp/files/9172577/ellipse.zip)

DomCR, 2022-07-23:
Hi @mattbrownfaro 

I've open a new branch with a fix, please check this PR https://github.com/DomCR/ACadSharp/pull/43 and confirm it works for your case.

closed 2022-07-25 by DomCR

mattbrownfaro, 2022-07-25:
Sorry I missed this comment. I will take a look at 43. Thank you!

DomCR, 2022-07-25:
This morning I publish a release in nuget https://github.com/DomCR/ACadSharp/releases/tag/v1.3.4-alpha

mattbrownfaro, 2022-07-25:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
>
> CAUTION: This email originated from outside the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.
>
> If you have clicked a link in this message with unexpected results, call IT immediately

Thank you Albert. I’ll get back to you later today.

mattbrownfaro, 2022-08-02:
Hi DomCR,

I have a ByLayer question. Is this how you are supposed to use it?

                    ACadSharp.Color color = dxfEnt.Color;
                    int idx = color.Index;
                    if (color.IsByLayer)
                    {
                          idx = dxfEnt.Layer.Color.Index;
                    }
The Sample_2018.dwg has a group of bylayer lines that are red in ACAD, but the Layer.Color.Index is 0.

***@***.***

Thanks,

Matt Brown | Sr Programmer
FARO Technologies, Inc.| Spokane, WA 99223
Mobile 509-995-2440
Nasdaq: FARO | ***@***.******@***.***> | www.faro.com<https://nam10.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.faro.com%2F&data=04%7C01%7Cmziegler%40inspired-tech.net%7C396be9692454496e050208d9cfc26c23%7C46c7c48fc6de455595c87a1742070915%7C0%7C0%7C637769256222737178%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=AMW80jNF2BOgUG2GKRGcvUDqOYJqvu%2FkbityWYl7u00%3D&reserved=0>

To receive FARO marketing material including but not limited to, Tech Sheets and Whitepapers, for new and existing FARO Products, please  Subscribe here<https://subscription.faro.com/> (you may select an alternative language).
To learn more about how FARO processes your personal data, please read our privacy policy which is available at:  Privacy Notice<https://www.faro.com/en-gb/privacy-policy>

Click here<https://subscription.faro.com/?rec=1&lang=en> to quickly and easily unsubscribe from receiving additional marketing material from FARO.
Unsubscribe<https://subscription.faro.com/?rec=1&lang=nl>       Se d?sabonner<https://subscription.faro.com/?rec=1&lang=fr>      Abmelden<https://subscription.faro.com/?rec=1&lang=de>      Disiscrizione<https://subscription.faro.com/?rec=1&lang=it>      Odmowa Subskrybcji<https://subscription.faro.com/?rec=1&lang=pl>      Cancelar la suscripci?n<https://subscription.faro.com/?rec=1&lang=es>      Cancelar a subscri??o<https://subscription.faro.com/?rec=1&lang=pt>      Oturumu kapat<https://subscription.faro.com/?rec=1&lang=tr>

**** This electronic mail message is transmitted exclusively for the intended recipient. It is not intended for transmission to, or receipt or use by, any unintended or unauthorized persons. Any review, use, printing, copying, retention, disclosure or distribution not expressly authorized is strictly prohibited and may violate applicable laws. If you have received this message in error, please (i) do not read it or any attachment, ii) immediately reply to the sender that you received the message in error, and (iii) erase or destroy the message and any copies that may exist. Thank you. ****