DomCR/ACadSharp #44: The class library generated an error

2420228300, 2022-07-24:
I can't run this instance project, ACadSharp class library in the various errors, is my use of the method wrong?

DomCR, 2022-07-24:
Hi @2420228300,

can you give some details about the error that you are getting, like a piece of code or the exception.

Thanks

2420228300, 2022-07-25:
[image]

DomCR, 2022-07-25:
This errors seem to be due a missing assembly, they all say that some part is not found or duplicated, check your project `ConsoleApp1` or create a new one that is clean and make sure that is well generated, then add the project reference.

The error `CS0579` says Duplicate 'attribute' attribute and is linked to `global::System.Runtime.Versioning.TargetFrameworkAttribute` this is a core assembly, you may have duplicated references to a framework.

2420228300, 2022-07-25:
Thank you very much for your answer I have successfully run the project, but how do I get all the text information in cad from the doc object? Is there any documentation for this?

2420228300, 2022-07-25:
I've found it, thank you very much

DomCR, 2022-07-25:
For more information here is a project with some examples:
https://github.com/DomCR/ACadSharp/tree/master/ACadSharp.Examples

closed 2022-07-25 by DomCR

2420228300, 2022-07-25:
ok