DomCR/ACadSharp #48: ByLayer color

mattbrownfaro, 2022-08-02:
I have a ByLayer question. Is this how it's used?
```
                    ACadSharp.Color color = dxfEnt.Color;
                    int idx = color.Index;
                    if (color.IsByLayer)
                    {
                          idx = dxfEnt.Layer.Color.Index;
                    }
```
The Sample_2018.dwg has a group of bylayer lines that are red in ACAD, but the Layer.Color.Index is 0.

 
[image]

DomCR, 2022-08-03:
Hi @mattbrownfaro

For all entities the default value for a color is `Color.ByLayer` but you are getting the `Entity.Layer` which it does not have the default setup, so is set as, `ByBlock` because the default value of an int is 0.

Check the `Color.cs` in the project files and you will notice that byBlock is set when the index is 0 and byLayer is set when index is 256.

```C#
/// <summary>
/// Defines if the color is defined by layer.
/// </summary>
public bool IsByLayer
{
	get { return this.Index == 256; }
}

/// <summary>
/// Defines if the color is defined by block.
/// </summary>
public bool IsByBlock
{
	get { return this.Index == 0; }
}
```

So, the entity is setup `ByLayer` but the layer itself, by default, is set as `ByBlock`

mattbrownfaro, 2022-08-03:
If the entity color is ByLayer, where do I find the correct color index? It wouldn't be in the entities.layer.color.index?

DomCR, 2022-08-04:
Correct, it would be there, but as I said the default for that is `ByBlock` which is indicated with the index being 0, so the 0 that you are getting is correct because the layer does not have any other color index assign.

mattbrownfaro, 2022-08-04:
Thanks Dom

closed 2022-08-05 by DomCR