DomCR/ACadSharp #49: Single Hatch entity DWG loads in True View but not in ACadSharp

mattbrownfaro, 2022-08-04:
Hi Dom,

When I read this DWG the BoundaryPath Edges and Entites are empty. I can't find any vector information in the hatch entity. It loads in True View and another cad package. Could you take a quick look and tell me if the data is there or not?

Thank you

[Hatch.zip](https://github.com/DomCR/ACadSharp/files/9263486/Hatch.zip)

DomCR, 2022-08-05:
Hi @mattbrownfaro,

I've looked at the file, it seems that there are some errors in the Hatch reading for dwg, I'll open a branch to solve this issue.

Thanks

mattbrownfaro, 2022-08-05:
> CAUTION: This email originated from outside the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.
>
> If you have clicked a link in this message with unexpected results, call IT immediately

Thank you Albert

closed 2022-08-08 by DomCR