DomCR/ACadSharp #52: Load performance

mattbrownfaro, 2022-08-18:
Hi Albert,

I'm running into a pretty big issue regarding load time. I have a 74K object count DWG that takes over 300 seconds to load in AS. Lots of small lines in this drawing. I could send you the DWG if interested.

Also, is there a timeline for when Polyline save will be implemented?

Thank you,
Matt

DomCR, 2022-08-19:
Hi @mattbrownfaro,

The low performance when reading a DWG may be due the flag `DwgReaderFlags.CheckCrc`, make sure that this flag is disabled in the `DwgReader`, by default is not set.

If the issue continue I'll open a branch to implement stress tests to explore the issue.

About the topic for write a Polyline, I'll open a dedicated branch to implement it.

DJGosnell, 2022-08-19:
> The low performance when reading a DWG may be due the flag `DwgReaderFlags.CheckCrc`, make sure that this flag is disabled in the `DwgReader`, by default is not set.

This was also my experience with loading files.

I feel like there are some other avenues I can pursue to increase the loading & saving performance, I just have to make some time.

mattbrownfaro, 2022-08-19:
Thank you Albert. I'll try out the flag and report back.

mattbrownfaro, 2022-08-19:
I had the flag set to none. I set it to CheckCrc just for curiosity and it was quite a bit slower as expected.

closed 2022-08-28 by DomCR