DomCR/ACadSharp #54: Layer names don't set correctly in `Entities` and `CadObject` for DXF files after using reader

eerFun, 2022-09-06:
Hi, I tried to read a `dxf` file by this package and distinguish every entity from it's layer name, but unfortunately all `Layer.Name` in `Entities` had the same value: `"0"` - the default layer name.
 I also tried to use `GetCadObject` by their `handler` but `cadObject`s had the same result, too.

Thanks for your attention. I’m looking forward to your reply.

DomCR, 2022-09-07:
Hi @eerFun,

Can you provide the version of the package, to make sure that it was not a version related issue.
I'll open a branch to make the test more accurate about layer related and work with a fix.

Unfortunately **I'll won't be available until October** to work in this issue, apologies for the delay.

eerFun, 2022-09-07:
Thank you for your quick response.

I was using the version `1.3.4-alpha`, but today I upgraded my package to the latest version `1.3.5-alpha` and unfortunately saw the problem still persists.

DJGosnell, 2022-09-07:
Can you provide a simplified DXF with this issue for review?

eerFun, 2022-09-08:
Yes, Of course.
[ImportCAD.zip](https://github.com/DomCR/ACadSharp/files/9525291/ImportCAD.zip)

closed 2022-09-30 by DomCR