DomCR/ACadSharp #62: Polygon contains an Arc Segment, wrong reading vertices

mahmoudadel54, 2022-11-28:
This is the drawn polygon in cad file.

[image]

And this is the output shape after reading it using ACadSharp:

[image]

and there is any indicator for the arc segment like [Center, startAngle, endAngle, Radius]

I think there is a bug in reading polygon contains arc because I tried to read an arc shape [a separate arc ] like this:
[image]

mahmoudadel54, 2022-12-05:
1- I have drawn a closed PolyLine starting with line [I drew 3 continious lines] and closed with ARC
so the result from AcadSharp is closed polygon from 4 lines so it ignored the ARC segment 

2- the second note is I tried to draw Pure ARC using arc command in cad and the result is some indicators that this is an ARC like center, radius ..etc

So the problem is when the ARC being part of a shape, it is ignored and if it is drawn separately, it is read well

DomCR, 2022-12-05:
The data seems to be there the problem is that Polyline vertices in Autocad create angles using this property:

```CSharp
/// <summary>
/// The bulge is the tangent of one fourth the included angle for an arc segment, made negative if the arc goes clockwise from the start point to the endpoint.A bulge of 0 indicates a straight segment, and a bulge of 1 is a semicircle
/// </summary>
[DxfCodeValue(DxfReferenceType.Optional, 42)]
public double Bulge { get; set; } = 0.0;
```

I'll open a branch linked to this issue to calculate the values of the center, radius and angle of an arc in a segment of a polyline.

closed 2022-12-08 by DomCR