DomCR/ACadSharp #66: System.ArgumentException

Setcad, 2022-11-29:
Hi, I get this type of error when to open the DWG file attached. Great project, keep it up please. Thanks.

System.ArgumentException: 'The item with the same key has already been added.'

gktval, 2022-11-29:
This is related to #2 here https://github.com/DomCR/ACadSharp/issues/63#issuecomment-1330626951. I'm glad you had an example for it. I agree that there should be a message about it, but that the file can continue to parse without an exception.

Setcad, 2022-11-29:
> This is related to (2) here [#63 (comment)](https://github.com/DomCR/ACadSharp/issues/63#issuecomment-1330626951). I'm glad you had an example for it. I agree that there should be a message about it, but that the file can continue to parse without an exception.

Yes I agree. I think DomCR can easly find the issue because of he is mastering the entire code algorithm

closed 2022-11-30 by DomCR