DomCR/ACadSharp #80: Angle Differences DXF & DWG

DJGosnell, 2022-12-08:
While working on DXF & DWG files in which the content is the same, I've noticed that the DWG reader reads angle values in radians (as documented)  I have a branch which shows these differences in MText, TextEntities, Arcs & Ellipses.

TextEntities show degrees on the DXF & Arc start and end angles.  All others are radians.  I was just going to make a PR, but I thought it would be worth discussing.  the DWG documentation rightfully states that the angle values will be in radians but the DXF is a little less explicit by stating "angle" for several types.

[netDXF](https://github.com/haplokuon/netDxf) handles this by ensuring all the angles are in degrees, but it seems that most of the angles in the files are actually in radians.  Do we want to just convert everything to make sure it's in radians during the reading process?

Angle DXF & DWG written in AutoCAD 2010 formats.
[angles.zip](https://github.com/DomCR/ACadSharp/files/10190564/angles.zip)

Output of the [dwg-dxf-angles branch](https://github.com/DJGosnell/ACadSharp/tree/dwg-dxf-angles):
```
DXF TextEntity Rotation: 45
DWG TextEntity Rotation: 0.7853981633974483
DXF Ellipse StartParameter: 1.020388469764473
DWG Ellipse StartParameter: 1.0203884697644738
DXF Ellipse StartParameter: 1.570796326794896
DWG Ellipse StartParameter: 1.5707963267948966
DXF Arc StartAngle: 45
DWG Arc StartAngle: 0.7853981633974483
DXF Arc EndAngle: 90
DWG Arc EndAngle: 1.5707963267948966
[angles.zip](https://github.com/DomCR/ACadSharp/files/10190563/angles.zip)

DXF MText Rotation: 0
DWG MText Rotation: 0
```

[image]

DomCR, 2022-12-09:
I'll keep them as **radians**, C# works with them so it's easier to work with `System.Math` functions if the angles are in radians.

DomCR, 2022-12-09:
My priority right now is to finish the first alpha for the `DwgWriter` I can open a branch dedicated to this issue but I'll not addressed for a while.

DJGosnell, 2022-12-09:
Excellent.  I'll issue a PR for this.

closed 2024-09-23 by DomCR