DomCR/ACadSharp #100: Insert Attibutes returning 'null' in Tag property

xaotix, 2023-02-06:
**Describe the bug**
All the insert Attributtes are returning null in Tag Property.

**To Reproduce**
Created a block with name 'M8_COM' with tags = 
'TRA_PEZ',
'DES_PEZ',
'NOT_PEZ',
'TIP_PEZ',
'ING_PEZ',
'COS_PEZ'

Insert the block and fill the properties.
Save the DWG as AutoCad 2010.

Tryed to read the block:
```
 using (DwgReader reader = new DwgReader(file))
                {
                    CadDocument acCurdb = reader.Read();
                    List<Insert> inserts = acCurdb.Entities.ToList().FindAll(x => x is ACadSharp.Entities.Insert).Select(x => x as ACadSharp.Entities.Insert).ToList();

                   forearch(var insert in inserts)
                   {
                    var atts = bloco.Attributes.ToList();
                   }
                 }
```

**Expected behavior**
When reading Tag property, return 'TRA_PEZ', 'DES_PEZ'...

**Screenshots**
[image]
[image]

**Additional context**
Obs.: I need the DWGs in 2010, for compatibility with others softwares.

DomCR, 2023-02-07:
Hi @xaotix,

This is caused by a missing implementation in the reader regarding the attributes, I'll open a branch to fix it.

xaotix, 2023-02-07:
@DomCR ok.

By the way, thank you and all the people involved in this amazing project!

DomCR, 2023-02-09:
There is a PR to fix this issue : https://github.com/DomCR/ACadSharp/pull/101
If you are using the repo please test that it has been fixed, if not I'll make a package release with the fix.

closed 2023-02-19 by DomCR