DomCR/ACadSharp #104: Load DWG 2018, then immediately save to DXF 2013 - issues

mattbrownfaro, 2023-02-27:
**Describe the bug**
I'm attempting to round trip a load/save in ACADSharp. From DWG2018 to DXF2013. Arcs are missing or flipped. Lost entity color.
Is there more I need to set up before saving dxf?

**To Reproduce**
            // See attached K6-2018.zip file
            string filenameDwg2018 = "C:\K6-2018.dwg";
            string filenameDxf2013 = "C:\K6-2013.dxf";
            
            CadDocument doc = ACadSharp.IO.DwgReader.Read(filenameDwg2018);
            doc.Header.Version = ACadVersion.AC1027;

            using (DxfWriter writer = new DxfWriter(filenameDxf2013, doc, false))
            {
                writer.Write();
            }

**Expected behavior**
I would like the DWG entities to save to DXF correctly.

**Screenshots**
[image]

If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
If the issue is related to an specific dwg/dxf file and if able, attach that file to the issue of an equivalent with the same error.
[K6-2018.zip](https://github.com/DomCR/ACadSharp/files/10842744/K6-2018.zip)

DomCR, 2023-02-28:
Hi @mattbrownfaro,

This problem is related to the current limitations of `ACadSharp`, the library is in a prerelease version and there are some features that are not finished yet, in this case is related to the limitations form the `DwgReader` and the `DxfWriter`. 

Your code is correct, if you want to have more information about what is happening you can use the event `OnNotification` in the `DwgReader` and `DxfWriter` so you can get notifications about the missing implementations or possible errors that may occur.

DomCR, 2023-02-28:
Exploring your file this is what the `DwgReader` cannot read, yet:
- MATERIAL
- ACDBDICTIONARYWDFLT
- MATERIAL
- VISUALSTYLE
- SUN
- SCALE
- TABLESTYLE
- MLEADERSTYLE

DxfWriter missing implementations:
- LwPolyline
- TextEntity
- Insert

DomCR, 2023-02-28:
I'll open different branches to implement this entities as soon as I can.

Also I'll open a branch for this issue to explore the arcs specification, it seems that the current writer don't get them right.

Thanks for your support!

mattbrownfaro, 2023-02-28:
> CAUTION: This email originated from outside the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.
>
> If you have clicked a link in this message with unexpected results, call IT immediately

You’re the best! We really are excited about using ACadSharp. It greatly simplifies things for us.

Thank you so much Dom!

Matt Brown | Sr. Software Engineer
FARO Technologies, Inc. | Spokane, WA 99223
Mobile 509.995.2440

Nasdaq: FARO | ***@***.***| www.faro.com

closed 2023-05-08 by DomCR