DomCR/ACadSharp #108: When the name of TEntity contains Chinese, garbled code appears in reading

3368aa, 2023-03-03:
[image]

DomCR, 2023-03-03:
Hi @3368aa 

can you post an example of file where this error occurs so I can reproduce?

thanks

3368aa, 2023-03-05:
send a email to u

DomCR, 2023-03-08:
`ACadSharp` reads the file with the right information, the problem seems to be with a configuration on windows that dos not allow to print Chinese characters in the console unless is configured that way.

Please check this reference link to see if this helps:
https://stackoverflow.com/questions/5750203/how-to-write-unicode-characters-to-the-console
https://stackoverflow.com/questions/36923984/c-sharp-windows-form-cannot-display-simplified-chinese-characters

Let me know if this works.

3368aa, 2023-03-12:
[image]

set Encoding UTF8

[image]

runtime value

[image]

output value

3368aa, 2023-03-12:
[image]
autocad 2020 display layers

3368aa, 2023-03-18:
[image]
[image]

DomCR, 2023-05-18:
I've found the error, the encoding is not implemented:
936                gb2312                       Chinese Simplified (GB2312)
The reader doesn't find the encoding and uses the default which does not support the Chinese characters.

I'll open a branch to implement the different encodings

closed 2023-06-08 by DomCR