DomCR/ACadSharp #110: Load specific types of entities only for read, improving the performance

xaotix, 2023-03-07:
**Is your feature request related to a problem? Please describe.**
No, it's a feature request.

**Describe the solution you'd like**
I want to load only specific type of entity, using less time to read the DWG.
The main goal mine is read the attributes from inserted blocks.

DomCR, 2023-03-08:
Hi @xaotix,

if this feature is quite complicated and I will have to investigate if is worth for time/resources wise, to give you some context about how the file reading works this is the minimal workflow to follow until you get to the objects:

`FileHeader` --> `DocumentHeader` --> `DxfClasses` --> `Handles` --> `Objects`

This is the minimum to get enough information to read the objects section. Once in the Objects Section we could test to add a filter by `ObjectType` or `DxfClass` considering that some objects go by one or the other (for example, the Hatch entity has an `ObjectType` but also a `DxfClass` depending on the version or the file itself). 

This seems easy but the problem is that in the `Handles` section, sometimes, is not complete, meaning that you find the different objects by reading their parent object, so to get the Attributes of an inserted block we may found some cases where they are missing because the handles are not there and you have to read the whole block to access to them.

Considering all this, we could investigate the amount of time and resources that we could spear by reading an specific `ObjectType` and without building the whole document.

xaotix, 2023-03-08:
@DomCR thank you for the clarification. 
I tried changing the code to only load the basic entities and the attributes. And it happened as you described: some objects returned missing or corrupted information.

DomCR, 2023-05-18:
Closing this issue due inactivity, please feel free to push a PR with a propose or open a discussion for further support.

closed 2023-05-18 by DomCR