DomCR/ACadSharp #113: How to detect draw order of objects

Nevors, 2023-03-16:
**Describe the bug**
I have two dwg files with difference draw order of two objects

**To Reproduce**
Parse two file and serialize to JSON. There is no difference except for modification date. They have the same draw order.

**Expected behavior**
I think these object should have different position in Objects array.

**Screenshots**
[image]
[image]

Correct me if i test it wrong.

Nevors, 2023-03-29:
@DomCR i tried to parse SORTENTSTABLE from files. As you can see in screenshot below, it looks like draw order.

Left side is the first screenshot from issue description, right is the second. 437 is green shape, 483 is blue shape.
[image]

What do you think of it? If it helps, i will push changes later.

DomCR, 2023-03-29:
now I see what you mean, `ACadSharp` should implement [this object](https://help.autodesk.com/view/OARX/2021/ENU/?guid=GUID-462F4378-F850-4E89-90F2-3C1880F55779) for that, by now this is not implemented but I can open a branch to implement this object into the library.

If you have a PR with an implementation feel free to push it into master and I'll be glad to review it, by now I'll open a branch to implement the `SORTENTSTABLE` object.

closed 2023-04-21 by DomCR