DomCR/ACadSharp #114: Can you add an example of block references in [ACadSharp.Tests]

8jxn, 2023-03-27:
In the example of [ACadSharp. Tests], it seems that there is only an example of creating a block. I am not sure how to create a block reference using this block. 
I think I need to use the [insert] class, but it doesn't work. Can you add a sample code for block references?

Thank you very very much

DomCR, 2023-03-27:
Hi @8jxn,

To give you some context, an `Insert` is an instance of a `BlockReference` stored in a document, that means that you need to assign this reference to the `Insert` to create the block that you want in the drawing, so to create a new block, with entities an all, you should create a new BlockReference in the table Blocks and to draw this block into the model you should create a new Insert into the Model that references this block.

Having said that I can see that the Insert class is incomplete, first of you should not be able to create an Insert without assigning a block to it. I'll create a branch to solve this issue.

DomCR, 2023-03-27:
Also, keep in mind that `ACadSharp.Tests` is meant to test the integrity of the code, you can check the project `ACadSharp.Examples` to see how it works, I'll add an example of this feature once the implementation is done.

8jxn, 2023-04-07:
Thank You Very Mach，IT is work now

closed 2023-04-12 by DomCR