DomCR/ACadSharp #115: Cannot read with AcadSharp dwg file created with AcadSharp

sv-ed, 2023-03-27:
**Describe the bug**
Cannot read with AcadSharp a dwg file written with AcadSharp. 

**To Reproduce**
I've created a dwg file with AcadSharp and then saved.
All the documents section (such as the Header) are as default.
When I try to load the file with `DwgReader` it throws the following exception: 
`ACadSharp.Exceptions.DwgNotSupportedException: 'Dwg version not recognised`.

Write dwg code:
```
// new document
ACadSharp.CadDocument doc = new ACadSharp.CadDocument();
// write file
ACadSharp.IO.DwgWriter writer = new ACadSharp.IO.DwgWriter("Test.dwg", doc);
writer.Write();
```
Read dwg code:
```
ACadSharp.CadDocument loaded = ACadSharp.IO.DwgReader.Read("Test.dwg");
```
Full exception and stack trace:
`ACadSharp.Exceptions.DwgNotSupportedException: 'Dwg version not recognised`
```
at ACadSharp.IO.DWG.DwgFileHeader.CreateFileHeader(ACadVersion version) in C:\Users\stage1\source\repos\ACadSharp\ACadSharp\IO\DWG\FileHeaders\DwgFileHeader.cs:line 37
at ACadSharp.IO.DwgReader.readFileHeader() in C:\Users\stage1\source\repos\ACadSharp\ACadSharp\IO\DWG\DwgReader.cs:line 225
at ACadSharp.IO.DwgReader.Read() in C:\Users\stage1\source\repos\ACadSharp\ACadSharp\IO\DWG\DwgReader.cs:line 97
at ACadSharp.IO.DwgReader.Read(String filename, DwgReaderConfiguration configuration, NotificationEventHandler notification) in C:\Users\stage1\source\repos\ACadSharp\ACadSharp\IO\DWG\DwgReader.cs:line 83
at ACadSharp.IO.DwgReader.Read(String filename, NotificationEventHandler notification) in C:\Users\stage1\source\repos\ACadSharp\ACadSharp\IO\DWG\DwgReader.cs:line 66
```

**Expected behavior**
No exception

**Screenshots**
From CadUtils:
[image]
It return `AcadVersion.Unknown`

DomCR, 2023-03-27:
Hi @sv-ed,

There is a dedicated test for this feature `ACadSharp.Tests.IO.DWG.DwgWriterTests` and it seems to work find, if that fails for you it may be related to the `Encoding` of the file.

If you open the file using a text editor can you see the version there?

[image]

sv-ed, 2023-03-27:
No, I cannot see it
[image]
How can I make sure to set the right encoding when saving the file?

DomCR, 2023-03-27:
This is strange... this part of the file is enforced in ASCII Encoding:

```C#
////0x00	6	“ACXXXX” version string
this._stream.Position = 0L;
this._stream.Write(Encoding.ASCII.GetBytes(this._document.Header.VersionString), 0, 6);
```

Which you should be able to read without any problem using a text editor.

Check the version of the header to be sure that the `CadDocument.Header.Version` is equal to `AC1018`.
Also if you could attach this corrupted file it would be great so I can take a look.

sv-ed, 2023-03-27:
`Header.Version` is set by default in `CadHeader.cs` as `ACadVersion.AC1018` and it was not modified.
Here is a screenshot from Quick Watch at this line: `ACadSharp.IO.DwgWriter writer = new ACadSharp.IO.DwgWriter("Test.dwg", doc);`
[image]

Link to the Test.dwg file below.
[Test.zip](https://github.com/DomCR/ACadSharp/files/11079175/Test.zip)

DomCR, 2023-03-27:
It seems that a part of the file is not written:
Your file:
[image]
My file:
[image]

You can se that both files are almost equal at the line `00000100` but the first 100 bytes are missing in your file and filled with empty bytes, the writer does that to save a space for the metadata:

```C#
public DwgFileHeaderWriterAC18(Stream stream, CadDocument model) : base(stream, model)
{
	// File header info
	for (int i = 0; i < this._fileHeaderSize; i++)
	{
		this._stream.WriteByte(0);
	}
}
```

The part that is missing is in this method, `DwgFileHeaderWriterAC18.writeFileMetaData`, if you are using this repo to debug, try to put a breakpoint at the beginning of this method to see if it hits, also check that your branch is fully updated with the current master branch.

DomCR, 2023-03-27:
So the string is written in the main stream... this part of the code uses the submodule `CSUtilities` to manage the streams, this should be up to date too, in older versions the stream was copied, if that's the case this could be writing the information somewhere else, please check the version of the submodule.

One last chance that you could try is to re-download the hole repo again in a separate folder and test the same code again, to make sure that everything is up to date.

If all this fail I'll need to explore this issue with more time.

sv-ed, 2023-03-27:
I can confirm that CSUtilities is at the [latest commit](https://github.com/DomCR/CSUtilities/tree/28a3ed81afa844c243f8d493c9dd06e3b29efa1e).

I tried cloning the AcadSharp repo again and then init and update sumbodule but unfortunately, same behaviour as described before.

DomCR, 2023-03-27:
try this:
```C#
using (DwgWriter writer = new DwgWriter("Test.dwg", doc))
{
    writer.Write();
}
```

sv-ed, 2023-03-27:
Well... yes, that worked, thank you :)

[image]

This also solves the issue that the stream wasn't closed successfully and the file resulted opened by another process, if I tried to open if after the write. 
Closing the program closed the stream too so the file was openable.

closed 2023-03-27 by sv-ed

DomCR, 2023-03-27:
perfect, I'll open a branch to apply a fix to manually dispose the stream to avoid this issue in the future.