DomCR/ACadSharp #117: Arcs not handeled correctly

sv-ed, 2023-03-27:
**Describe the bug**
The arc entities are not handled correctly throughout the AcadSharp library.
All the files contain 2 arcs with the same starting point and bulge but with different end point:
- arc 1 with points (100,100), (200,200) and bulge = 1
- arc2 with points (100,100), (150,150) and bulge = 1

The files **should** have the same entities with the same values.
- Input.dwg is drawn with AutoCad
- Output.dxf contains the read Input.dwg file written with AcadSharp
- Output2.dxf contains the arcs generated by AcadSharp

**To Reproduce**
Example 1:
- Read with `DwgReader` Input.dwg
- Write Output.dxf with `DxfWriter`
- Observe the difference between the 2 files

Code:
```
ACadSharp.CadDocument loaded = ACadSharp.IO.DwgReader.Read("Input.dwg");
ACadSharp.IO.DxfWriter.Write("Output.dxf", loaded, false);
```

Example 2:
- Create a new `CadDocument`
- Create 2 arcs
- Write Output2.dxf with `DxfWriter`

Code:
```
ACadSharp.CadDocument doc = new ACadSharp.CadDocument();
doc.Entities.Add(ACadSharp.Entities.Arc.CreateFromBulge(new CSMath.XY(100, 100), new CSMath.XY(200, 200), 1));
doc.Entities.Add(ACadSharp.Entities.Arc.CreateFromBulge(new CSMath.XY(100, 100), new CSMath.XY(150, 150), 1));
ACadSharp.IO.DxfWriter.Write("Output2.dxf", doc, false);
```

**Expected behavior**
I expect the two Output files to be as the Input one.

**Additional context**
[Files.zip](https://github.com/DomCR/ACadSharp/files/11079856/Files.zip)
AcadSharp up to date with master branch at b17c7dfdf9abc12716583a37a505e4a5cc861219

DomCR, 2023-03-27:
There is a similar issue related to the arcs:
https://github.com/DomCR/ACadSharp/issues/104

Branch assigned to fix this one too.

closed 2023-05-08 by DomCR

sv-ed, 2023-05-09:
Thanks for addressing the issue, now it's working correctly :)