DomCR/ACadSharp #127: Argument out of range exception - will not load

mattbrownfaro, 2023-05-15:
**Describe the bug**
The attached dxf loads in True View, but throws and exception on a DxfReader.Read()

**To Reproduce**
Try loading the DXF

*Screenshots**
[image]

[out of range error.zip](https://github.com/DomCR/ACadSharp/files/11480906/out.of.range.error.zip)

Thank you Dom!

DomCR, 2023-05-16:
Hi @mattbrownfaro 

The DXF has some default values set to 0, which is invalid for dates, I'll add a try catch to avoid this exception and let the exception be handled by the notification system in the header setter.

mattbrownfaro, 2023-05-16:
> CAUTION: This email originated from outside the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.
>
> If you have clicked a link in this message with unexpected results, call IT immediately

That’s awesome, I’ll check it out.

I just want to say…so much faster, and object support getting so close. Couldn’t be happier or more impressed with your work.

Matt Brown | Sr. Software Engineer
FARO Technologies, Inc. | Spokane, WA 99223
Mobile 509.995.2440

Nasdaq: FARO | ***@***.***| www.faro.com

mattbrownfaro, 2023-05-16:
Dom,

It no longer crashes, but I get zero objects loaded too. Is there a way to just catch the error and continue. The file loads fine in True View and other packages.

Thanks!

Matt

DomCR, 2023-05-16:
Just for curiosity, which software is the creator of this dxf file? the format seems quite different from the one specified in the documentation.

mattbrownfaro, 2023-05-16:
It was sent in by a customer and I didn’t get that information. Sorry, I know the DWG format can be a mess.

As a test I will resave it in ACAD and see if it loads, and will get back to you.

Thanks!

mattbrownfaro, 2023-05-16:
I converted the dxf to dwg using this online utility: https://cloudconvert.com/dwg-converter

I normally use True View but it doesn’t support converting DXF to DWG.

It loads file after converting.

CAUTION: This email originated from outside the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

If you have clicked a link in this message with unexpected results, call IT immediately

Just for curiosity, which software is the creator of this dxf file? the format seems quite different from the one specified in the documentation.

closed 2023-05-18 by DomCR

mattbrownfaro, 2023-05-18:
Hi Dom - I pulled master and rebuil. There are still no objects after reading "out of range error.dxf". Should I be getting the objects now?  Thanks

DomCR, 2023-05-18:
Hi @mattbrownfaro,

So the Model_Space* has been found for this type of dxf I decided to merge it with master the keep the scope of the task.

Now the problem is with the entities, they don't have the handle of the owner which doesn't make much sense because we need to know in which block belong, here is a sample of the test log:

Owner [missing handle] not found for ACadSharp.IO.Templates.CadHatchTemplate with handle 133
Owner  not found for ACadSharp.IO.Templates.CadLwPolylineTemplate with handle 135
Owner  not found for ACadSharp.IO.Templates.CadLwPolylineTemplate with handle 136
Owner  not found for ACadSharp.IO.Templates.CadLwPolylineTemplate with handle 137
Owner  not found for ACadSharp.IO.Templates.CadEntityTemplate with handle 138

I need to explore this document because the format doesn't match the documentation and I would like to perform some tests at the document to see how other frameworks or Autocad itself handles this kind of misinformation.

I'll open another branch to check this issue, worst case scenario I'll add an option to the reader to assign all the entities without owner to the model_space* as a default owner, which I'm guessing is what Autocad will do.

mattbrownfaro, 2023-05-18:
Hi Dom,

Unfortunately, there are many undocumented flavors of DXF. I’m not too concerned about this drawing; it’s probably an outlier. Would it be helpful to have the drawing it converted to, which loads?

Hi @mattbrownfaro<https://github.com/mattbrownfaro>,

So the Model_Space* has been found for this type of dxf I decided to merge it with master the keep the scope of the task.

Now the problem is with the entities, they don't have the handle of the owner which doesn't make much sense because we need to know in which block belong, here is a sample of the test log:

Owner [missing handle] not found for ACadSharp.IO.Templates.CadHatchTemplate with handle 133
Owner not found for ACadSharp.IO.Templates.CadLwPolylineTemplate with handle 135
Owner not found for ACadSharp.IO.Templates.CadLwPolylineTemplate with handle 136
Owner not found for ACadSharp.IO.Templates.CadLwPolylineTemplate with handle 137
Owner not found for ACadSharp.IO.Templates.CadEntityTemplate with handle 138

I need to explore this document because the format doesn't match the documentation and I would like to perform some tests at the document to see how other frameworks or Autocad itself handles this kind of misinformation.

I'll open another branch to check this issue, worst case scenario I'll add an option to the reader to assign all the entities without owner to the model_space* as a default owner, which I'm guessing is what Autocad will do.