DomCR/ACadSharp #131: Invalid dxf code with value 0

ChristopherVR, 2023-05-17:
**Describe the bug**
I'm unsure as to why any DxF file I create using SysCAD causes an exception to be thrown when trying to open it in this library.

**To Reproduce**
Try to read the DxfFile I've attached.

**Expected behavior**
An exception should not be thrown as I'm still able to open this file in other .NET DxfFile parsers. 

**Screenshots**
[image]

**Additional context**
[05_Flowsheet.zip](https://github.com/DomCR/ACadSharp/files/11493136/05_Flowsheet.zip)

DomCR, 2023-05-18:
Hi @ChristopherVR 

It seems that SysCAD **does not organize** the file using `Subclasses`, unfortunately the current state of this project does not allow this kind of dxf format, the intention is to change the mapper that currently assigns the values by dxf class to make the parser compatible with this format but that will require a lot of code because all the current readers must be changed.

I'll open a branch to explore alternatives or to start with this development.

DomCR, 2023-07-11:
The current Dxf reader can read this kinds of files, but I've noticed that they are incomplete, there are missing elements and the result file is an incomplete `CadDocument` that cannot be saved, I'll link this issue to the 2 branches that have to fix this issue.

closed 2024-06-14 by DomCR