DomCR/ACadSharp #134: Simple POINT/TEXT DWG only loads a single point

mattbrownfaro, 2023-05-22:
**Describe the bug**
This DWG should have 200+ objects. Only returns a single point from ACadSharp

**Expected behavior**
Read should return Points and Text labels, and a polyline or 2

**Screenshots**
This is how it looks in True View
[image]

**Additional context**
I ran it thru ConvertCloud.com and it loads correctly in ACadSharp. 

[23 wheeling.zip](https://github.com/DomCR/ACadSharp/files/11532199/23.wheeling.zip)

DomCR, 2023-05-24:
Hi @mattbrownfaro,

Looking at the file I've found a bug in the `ObjectReader` pushing the changes now.

closed 2023-05-24 by DomCR

mattbrownfaro, 2023-05-24:
Fixed!!! Well done Dom! Thank you.