DomCR/ACadSharp #138: Insert entity export

Revan1985, 2023-06-06:
Dear all,
I am using this library to read some dwg files.
I have been able to read almost all the entities in the file, but I am having issues with the Insert Entities.
What I need is to understand how to position entities in the Insert type correctly in the world.
Now seems that all related entities are positioned well, but rotated some kind of 90 degrees

`
            EntityTransform transform = new()
            {
                Rotation = (float)insert.Rotation,
                Scale = new((float)insert.XScale, (float)insert.YScale),
                Translation3d = new((float)insert.InsertPoint.X, (float)insert.InsertPoint.Y, (float)insert.InsertPoint.Z),
            };

            transforms_[transformCount] = transform;

            foreach (Entity entity in insert.Block.Entities)
            {
                if (insert.Normal.Z == -1)
                {
                    transform.Rotation *= -1;
                    transform.Scale = new(-(float)insert.XScale, (float)insert.YScale);
                    transform.Translation3d = new((float)-insert.InsertPoint.X, (float)insert.InsertPoint.Y, (float)insert.InsertPoint.Z);
                }
                ExportEntity(entity, layer, transforms_);
            }`
I am using the quoted code, where insert is my insert entity.
I am transforming the entities by father, but rotation is bugged (and some entities are pixels away from their position).
Which is the right code to calculate it? I am not able to find any useful example online

The following images show result, first one is from Autodesk DWG TrueView
[image]
this one is from my code...
[image]

Thank you for any help, will be appreciated

DomCR, 2023-06-08:
Hi @Revan1985,

There is a couple of layers to understand about the `Insert` entity, this is directly related to the `BlockRecord` that is referencing, this is how it works:

**Insert**: instance of a block record in the **model**, the fields like `InsertPoint`, `Scale` and `Rotation` are modificators for the actual block that is referencing which will be applied to the drawing of the instance keeping the entities in the block intact.

**Insert.Block**: this is the block which the insert is referencing to, this must exist in the document and contains all the entities, in the block you can find the `BlockEntity` which contains the base information for the block, the one that may interest you is the `BasePoint` which specifies the origin point for the block. 

I'll try to create some examples and do some testing for you issue, this what the documentation says about how the insert works but Autodesk is quite messy with the docs and sometimes are not accurate enough, I'll reach out again with a code example.

DomCR, 2023-06-08:
A note for the rotation, keep in mind that the rotation is in **radians**, and make sure to update your branch/package to the last version, not long ago there was a fix to write and read the angles for dxf.

Revan1985, 2023-06-09:
OK, thank you for the answer.
I could have understood what I was missing.
I will update this issue as soon as I test it.

Revan1985, 2023-06-09:
Thanks to your answer I resolved near 90% of my problem.
Now I am getting some wrong side rotations, and positions. 
I have something more to check before rendering?

Changing my code from original one to this

`
int transformCount = transforms.Count();
EntityTransform[] transforms_ = transforms.ToArray();
Array.Resize(ref transforms_, transformCount + 1);

EntityTransform transform = new()
{
    Rotation = (float)insert.Rotation,
    Scale = new((float)insert.XScale, (float)insert.YScale),
    Translation3d = new((float)insert.InsertPoint.X, (float)insert.InsertPoint.Y, (float)insert.InsertPoint.Z),
};

transforms_[transformCount] = transform;
// if normal is -1, rotate 180°
if (insert.Normal.Z == -1)
{
    transform.Rotation *= -1;
    transform.Scale = new(-(float)insert.XScale, (float)insert.YScale);
    transform.Translation3d = new((float)insert.InsertPoint.X, (float)-insert.InsertPoint.Y, (float)insert.InsertPoint.Z);
}

ExportBlockRecord(insert.Block, insert.Layer, transforms_);
`
`private void ExportBlockRecord(BlockRecord block, Layer layer, IEnumerable<EntityTransform> transforms)
{
    int transformCount = transforms.Count();
    EntityTransform[] transforms_ = transforms.ToArray();
    Array.Resize(ref transforms_, transformCount + 1);

    Vector3 basePosition = block.BlockEntity.BasePoint.ToVector3();

    EntityTransform baseTransform = new()
    {
        Translation3d = basePosition.ToPoint3d(),
    };

    transforms_[transformCount] = baseTransform;

    foreach (Entity entity in block.Entities)
    {
        ExportEntity(entity, layer, transforms_);
    }
}`

just work near perfection. I don't understand if something is wrong, and what...
cause 9 entity on 10 is right positioned and last one no...
[image]

you can see an example here.
Sink are right, but lower one is rotated, just like right wc.

Thank you again for your patience.

DomCR, 2023-06-11:
Outisde `ACadSharp` I'm quite clueless but if I have to guess I would suggest to review this piece of code:

```C#
if (insert.Normal.Z == -1)
{
transform.Rotation *= -1;
transform.Scale = new(-(float)insert.XScale, (float)insert.YScale);
transform.Translation3d = new((float)insert.InsertPoint.X, (float)-insert.InsertPoint.Y, (float)insert.InsertPoint.Z);
}
```

By changing the normal you are applying a symmetry using the plane XY, the rotation that had in the first state may be to correct the difference between the insert point and the position of the object, by rotating the element again this may cause the invalid positions in your drawing. Try to isolate the elements that are badly rotated and check if they go inside the if and try to correct the rotation by changing the `transform.Rotation *= -1;`

Hope it helps!

Revan1985, 2023-06-14:
I have done an export between my old cad exporter and the one using acad.
There is something that I don't understand, I am exporting entities grouping by ObjectType, and in some case, i have the following difference
[image]
There are some block reference that are readed in this way.
The right one is the old exporter, left one is AcadSharp.
This could be an error in reading the dwg file? Or I am just drunk trying to export in the right way? 🤣 
Anyway, I have noticed that the exporting is reading in the same way the wrong rotated entities, so this is my error in the export, I will figure what I have missed...

Thank for your pacience 😃

DomCR, 2023-06-14:
The order of how `ACadSharp` or any framework reads the different objects should not matter, Autocad creates a section in the dwg file that should contain all the Handles (unique object id) and the offset of where the object is, but in practice the some objects only gives the parent, for example a block, and when you read the block you get all the entities in it. depending on how the framework stores this handles the reading order will vary but the information at the end should be the same.

In the image the values are the same so I'm guessing that everything went fine (I cannot see the angle) so I wouldn't worry about the order.

Hope that this information helps, as I said in the other message your problem may be related on how you are handling the symmetry of the objects with an inverted normal, try to isolate those and see what's missing.

Revan1985, 2023-06-15:
I have done an export writing all the entities to text file.
I am now sure issue is not the library, but my exporting code, since entities are exactly identical from Acad to old reader (except some angles, but this is related to how cad set them, and old exporter calculates, but they are correctly used now).
I will try to check my rendering code, and see what I have missed... 😭 

Thank you for your precious support 😺

Revan1985, 2023-08-04:
Hello again.
I have resolved almost all the issue with the exporting code (there was a mistake between the old library and new one for transformation).
Using this code

`            Vector2 p = original;
            p *= scale;
            double theta1 = Math.Atan2(p.X, p.Y);
            double theta2 = theta1 + rotation;
            double r1 = p.Length(); // Math.Sqrt(p.X * p.X + p.Y * p.Y);
            p.X = (float)(r1 * Math.Cos(theta2));
            p.Y = (float)(r1 * Math.Sin(theta2));
            p += translation;
            return p;`

instead of this one
`System.Numerics.Matrix4x4 sc = System.Numerics.Matrix4x4.CreateScale(scale.X, scale.Y, 1f);
            System.Numerics.Matrix4x4 ts = System.Numerics.Matrix4x4.CreateTranslation(translation.X, translation.Y, 0.0f);
            System.Numerics.Matrix4x4 rt = System.Numerics.Matrix4x4.CreateRotationZ((float)rotation);

            System.Numerics.Vector2 v = System.Numerics.Vector2.Transform(new System.Numerics.Vector2(original.X, original.Y), sc);
            v = System.Numerics.Vector2.Transform(v, rt);
            v = System.Numerics.Vector2.Transform(v, ts);
            return new(v.X, v.Y);`
did not transform in the right way with Acad Sharp, so I switched to numerics matrix.
Thank you for your precious support and this wonderfull library (there are still some transform mistake, but I think is how i read them again)

closed 2023-08-04 by Revan1985