DomCR/ACadSharp #141: Entities are not read correctly in specific DWG file

eugeniobidone, 2023-06-12:
**Describe the bug**
In the attached DWG file the program finds only one geometric entity instead of the around 200 expected entities.

**To Reproduce**
Use the DwgReader.Read() function to read the attached DWG file obtaining the corresponding CadDocument object.

**Expected behavior**
The CadDocument object obtained from the attached DWG file should contain around 200 entities, not only one.

[silenziatore per oriano.zip](https://github.com/DomCR/ACadSharp/files/11718994/silenziatore.per.oriano.zip)
[image]

DomCR, 2023-06-12:
Hi @eugeniobidone,

I've tested your file and I got 202 entities in the **ModelSpace**

Please check that you are using the latest Package in Nuget or update your branch with the latest changes, if the error persist please share the piece of code that you are using.

eugeniobidone, 2023-06-12:
Hi @DomCR ,

I am using ACadSharp package in an Azure function, and when I run it in debug mode I obtain 1 entity in the ModelSpace. 

I have now tested the same file in a Console App, and I obtained 202 entities as expected.

Furthermore, the same Azure function after deployment reads 202 entities, so the problem appears only when I work in debug mode with my Azure function.

I don't know why it happens, but the important thing is that it works fine after deployment.

Thank you for your quick answer

closed 2023-07-25 by eugeniobidone