DomCR/ACadSharp #142: DXF loads much better once processed by Cloud Convert

mattbrownfaro, 2023-06-12:
**Describe the bug**
Loading the attached DXF has a lot of errors and get pretty bad results. If I run it thru Cloud Convert and reload it, it looks great.

**To Reproduce**
Unzip and load ALFA Sportwagon-4 views.dxf - I get lots of circular arcs
Convert it in (https://cloudconvert.com/) and it looks great

**Expected behavior**
Should get 4 nice views of the wagon

**Screenshots**
[image]

**Additional context**
With most of my dxf/dwg files, I have no idea where they came from.

[ALFA Sportwagon-4 views.zip](https://github.com/DomCR/ACadSharp/files/11728772/ALFA.Sportwagon-4.views.zip)
[ALFA Sportwagon-4 views-CloudConvert.zip](https://github.com/DomCR/ACadSharp/files/11728782/ALFA.Sportwagon-4.views-CloudConvert.zip)

DomCR, 2023-06-14:
Hi @mattbrownfaro,

The `DxfReader` is hardly coded to work only with the Autocad standard, which I'm seeing now that is an error. It seems that dxf is a pretty flexible format so I'm working on a full refactor for the reader/writer to make it work.

I've assigned the PR for the [DxfReader](https://github.com/DomCR/ACadSharp/pull/143) for this issue.

Thanks for the files, I'll use them for testing.

closed 2023-07-11 by DomCR

mattbrownfaro, 2023-07-31:
Hi @DomCR,

Looking MUCH better but missing objects. And throwing exceptions.
[image]
Missing wheels and other views of the car, but geometry looks great.

This drawing is throwing a lot of exceptions. I believe it loaded in a previous ACADSharp version
[2140QUEENCITY.zip](https://github.com/DomCR/ACadSharp/files/12222797/2140QUEENCITY.zip)

Built DxfWriter-Full-refactor

Thanks Dom!