DomCR/ACadSharp #155: Find TDCREATE from the header of DWG

amazingProj, 2023-07-31:
Hello

While I was taking files that have both .dwg and .dxf and extracting the header of the files by this code, I notice that the one in the header that should be TDCREATE is actually the TDUCREATE.

Could you help me gets the TDCREATE from the header?

Thanks

DomCR, 2023-08-07:
Hi @amazingProj 

I've been exploring different files using the `DwgReader` and the TDCREATE seems to match with the desired value. Could you provide some examples or images about this issue?

TDUCREATE is not stored in the dwg format, I'm looking for some documentation to find out where this field is created/processed, but I think is referenced to the TDCREATE using the Universal date/time instead of the Local date/time.

To find the TDCREATE you only need to read the file and go the the path `CadDocument.Header.CreateDateTime`

amazingProj, 2023-08-07:
Sorry, I was wrong.
I have discovered that AutoCAD adds TDCREATE which is the convertion to local time of the original DWG time based on your timezone.
 AutoCAD takes TDCREATE of DWG and names it TDUCREATE.

closed 2023-08-07 by amazingProj