DomCR/ACadSharp #156: ACadSharp on Unity

Kanhoto, 2023-08-09:
**Describe the bug**
It's not really a functionnal bug, but when I'm .net 2.1 dll in my Unity project, I can use it to read files (although I'm still a beginner with dwg files, but I suppose it's fine) but I get this error about a function in ACadSharp

**To Reproduce**
* Opening a Unity project
* Adding ACadSharp using NugetForUnity
* (Zip open the nupkg, to extract the dll for Unity)
* And you can see the red line in Unity Console

**Expected behavior**
To not have an error

**Screenshots**
[image]

**Additional context**
I can't add it, but I used the sample_AC1014.dwg, but it doesn't matter since it's not the issue

DomCR, 2023-08-09:
Hi @Kanhoto 

The .net2.1 has a dependency with:
```xml
  <ItemGroup Condition="'$(TargetFramework)' == 'netstandard2.1'">
    <PackageReference Include="System.Memory" Version="4.5.5"/>
    <PackageReference Include="System.Text.Encoding.CodePages" Version="7.0.0" />
  </ItemGroup>
```
For the first issue in the image is asking for the `System.Memory` library, not sure if you will need to add the `System.Text.Encoding.CodePages` I think that the current version in the nuget package is not updated yet. Try to add this to you project and check if you can run it.

For the second error, it says that "this assembly can cause crashes in the runtime" this may be due it's marked as a pre-release or is not a Unity package.

For more information about this you can check the changes added in this https://github.com/DomCR/ACadSharp/pull/93/files

Let me know if this has resolved your issue, thanks.

Kanhoto, 2023-08-09:
Hi, thanks for the answer,

I switched to net48 dll, disabled the assembly verification in player settings and I don't have these errors anymore.
Just, do you have a list of compatible object ? I'm trying this lib and netDxf, I included both since this one can take DWG, but for DXF, I dunno which one works better.

https://github.com/haplokuon/netDxf

DomCR, 2023-08-10:
`ACadSharp` supports dxf, check `ACadSharp.IO.DxfReader`

Unfortunately there is not a documentation page for this library yet.

I think that `netDxf` is in full release, meaning that maybe can provide more stability, `ACadSharp` is still in Alpha, which means that some of the features are not fully implemented and might be a little buggy.

If you have any issue using `ACadSharp` please add an issue to this repo and I'll address it as soon as I can.

Kanhoto, 2023-08-10:
Well I'm trying to redraw either dwg or dxf in Unity, to provide a viewer, do you have any clue how I could get all the necessary data ? like for lines, to get their length and 2 vertices ?

DomCR, 2023-08-20:
All the "drawable" entities are in the `CadDocument.Entities` this list provides all the objects contained in the Model Space of the drawing, each one has it's own fields defining the geometry, for example a line has `Line.StartPoint` and `Line.EndPoint` to define it's own geometry. You should process each `Entity` on it's own.

Kanhoto, 2023-08-28:
[image]
https://all3dp.com/2/best-free-dwg-viewers-autocad/

Am I taking risk using ACadSharp or it's safe ?

DomCR, 2023-08-28:
I didn't know about these lawsuits, but I think they sued the developers when they use the dwg format in a commercial use, I've found this article that talks about this, in the article there are multiple references to different open-source projects that read and write dwg files:
https://en.wikipedia.org/wiki/.dwg

Also, in case of ACadSharp, I've haven't used revere engineering to read the file, the format specification is publish by the OpenDesign https://www.opendesign.com/files/guestdownloads/OpenDesign_Specification_for_.dwg_files.pdf 

As far as I know is not illegal to use or implement ACadSharp, but I don't know about using the library in a commercial use.

Kanhoto, 2023-08-31:
I see that ACadSharp take DWF format, so I should be able to do what I want, even for commercial use.
Thanks for your help (even though I drifted from the original subject ^^)

closed 2023-08-31 by Kanhoto

Kanhoto, 2023-08-31:
> I didn't know about these lawsuits, but I think they sued the developers when they use the dwg format in a commercial use, I've found this article that talks about this, in the article there are [...]

You should maybe add a line in your README, because your package is distributed with "MIT license" and some people may misunderstand that. You won't be responsible, but it just to prevent people from making that mistake (^^;)