DomCR/ACadSharp #163: Still throwing exceptions.

mattbrownfaro, 2023-09-06:
Hi @DomCR,

Looking MUCH better but missing objects. And throwing exceptions. [ALFA Sportwagon-4 views.zip](https://github.com/DomCR/ACadSharp/files/11728772/ALFA.Sportwagon-4.views.zip)
[image]
Missing wheels and other views of the car, but geometry looks great.

This drawing is throwing a lot of exceptions. I believe it loaded in a previous ACADSharp version
[2140QUEENCITY.zip](https://github.com/DomCR/ACadSharp/files/12222797/2140QUEENCITY.zip)

Built DxfWriter-Full-refactor

Thanks Dom!

_Originally posted by @mattbrownfaro in https://github.com/DomCR/ACadSharp/issues/142#issuecomment-1659305567_

DomCR, 2023-09-11:
Hi @mattbrownfaro 

The branch [DxfWriter-Full-refactor](https://github.com/DomCR/ACadSharp/tree/DxfWriter-Full-refactor) is currently in development, all the exceptions are expected, when the task is completed and merge this should work fine the **ALFA Sportwagon-4 views** file.

About the **2140QUEENCITY** is more complicated, the dxf writer that they used encoded the name of the model space in a funny way _$MODEL_SPACE_ which messes up the way that `ACadShrap` finds it.

About the missing wheels, it seems that they are drawn with polylines, for some reason the ones that the writer is generating are not valid.

I'll open 2 branches to take a look to both issues.

DomCR, 2023-10-27:
This issue seems to be bigger than I expected, the file that is failing is because is in an older version `AC1009` which is not supported by `ACadSharp` at the moment, by now I'll add an exception to the writer to indicate that this version is not supported but it will take time do make the `DxfReader` compatible with it.

closed 2024-07-02 by DomCR