DomCR/ACadSharp #173: Some entities cannot be shown in CAD file after created. Opening CAD needs to be repaired.

TzhongWei, 2023-09-26:
When I created Polyline2D and LwPolyline, the entities could not be displayed in the dwg file. Besides, after creating the file, AutoCAD always asked me to repair it. After repairing the dwg file, the layer I set couldn't function. I mean the layer I only saw "0" and "$RECOVER_230926185604-3".
I used the student 2020 version. Is it because of the version? 
I've not tried other entities, but so far Point and Line can work.

Here is the program

 public void string writeCADFile(string FileStream)
        {
            ACadSharp.CadDocument doc = new ACadSharp.CadDocument();
            var NewLayer = new ACadSharp.Tables.Layer("TestLayer");
            NewLayer.Color = new ACadSharp.Color(0,0,0);

            ACadSharp.Entities.Polyline2D polyline = new ACadSharp.Entities.Polyline2D();
            List<ACadSharp.Entities.Vertex2D> vertices = new List<ACadSharp.Entities.Vertex2D>()
            {
                new ACadSharp.Entities.Vertex2D(){ Location = new XYZ(0,0, 0)},
                new ACadSharp.Entities.Vertex2D(){ Location = new XYZ(1,1, 0)},
                new ACadSharp.Entities.Vertex2D(){ Location = new XYZ(2, 2, 0)}
            };
            polyline.Vertices.AddRange(vertices);
            
            doc.Entities.Add(polyline);
            using (DwgWriter writer = new DwgWriter(FileStream, doc))
            {
                writer.Write();
            }
        }

TzhongWei, 2023-09-27:
Besides, I tried the following example to copy Entities from A.dwg to B.dwg. The new dwg showed errors leading it not to be opened when I used AutoCAD to read it.
public static void DwgEntitiesToNewFile(string input, string output)
        {
            CadDocument doc = DwgReader.Read(input);

            //New document to transfer the entities
            CadDocument transfer = new CadDocument();

            //Move the entities to the created document
            List<ACadSharp.Entities.Entity> entities = new List<ACadSharp.Entities.Entity>(doc.Entities);
            foreach (var item in entities)
            {
                ACadSharp.Entities.Entity e = doc.Entities.Remove(item);
                transfer.Entities.Add(e);
            }

            //Save the document
            using(var writer = new DwgWriter(output, transfer))
            {
                writer.OnNotification += NotificationHelper.LogConsoleNotification;
                writer.Write();
            }
        }

DomCR, 2023-09-28:
Hi @TzhongWei

About the `DwgWriter` please read the documentation [here](https://github.com/DomCR/ACadSharp#dwg-writer) if you want to save a change in your file I would recommend to use the `DxfWriter` instead, which is more stable and reliable.

DomCR, 2024-09-23:
`DwgWriter` is completed and working, closing this issue.

closed 2024-09-23 by DomCR