DomCR/ACadSharp #178: Reading some dwg-files results null values for CadDocument.ModelSpace and PaperSpace

thorster, 2023-10-13:
**Describe the bug**
Great work with ACadSharp! Just recently started to use it to read dwg-files and extract text from block attributes, All worked fine for the first batch of 600+ dwg-files. Now with a second batch something goes wrong. The CadDocument properties ModelSpace and PaperSpace are both null and a null-reference exception is thrown when my code access the property Entities.

**To Reproduce**
This code should reproduce the issue:

                using var reader = new DwgReader("691501_14101.dwg");
                var doc = reader.Read();
                var entities = doc.Entities;

The dwg-file:
[691501_14101.zip](https://github.com/DomCR/ACadSharp/files/12894427/691501_14101.zip)

**Expected behavior**
The dwg-files are created in BricsCAD. I have opened a dwg-file in AutoCAD and used the command AUDIT but no errors where found. Saved the file in AutoCAD but still same behaviour in ACadSharp. I have verified that the files can be read by AutoCAD and BricsCad .NET APIs. I have updated ACadSharp from Nuget today.

**Screenshots**

[image]

DomCR, 2023-10-14:
Hi @thorster,

Debugging the file I've found that there was a wrong assignment for a block flag, I've opened a branch and apply the fix to master, I'll make a Nuget release shortly.

closed 2023-10-14 by DomCR

thorster, 2023-10-14:
Excellent! I will test later today. Much apreciated 😀

thorster, 2023-10-15:
The fix solved the problem. Thanks!