DomCR/ACadSharp #182: The item with handle # is already assigned to a document

thorster, 2023-10-15:
**Describe the bug**
When reading a few dwg-files an exception is thrown: "The item with handle # is already assigned to a document". The exception is thrown in CadDocument.addCadObject
**To Reproduce**
Read this file ;
[1TSX 265484-YBG100.zip](https://github.com/DomCR/ACadSharp/files/12909153/1TSX.265484-YBG100.zip)

thorster, 2023-10-15:
Also found another perhaps related issue. This exception: "An item   with the same key has already been added. Key: 131"

You should be able to reproduce with this file

[691501_3969-1.zip](https://github.com/DomCR/ACadSharp/files/12909193/691501_3969-1.zip)

DomCR, 2023-10-15:
Hi @thorster 

I've been looking at the files, they have 2 different issues:

First file: the assignment of the `RootDictionary` in the `CadDocument` is wrong which causes an error when is assigned for a second time, it seems that some data is missing or misplaced because some objects don't have the owner written in the file which is odd.

Second file: there is a duplication for a `ReactorHandle` similar as before, when the second reactor is added this causes an error, this can be solved easy.

I'll open a branch to take care of this issues, thanks for reporting!

closed 2023-10-16 by DomCR