DomCR/ACadSharp #184: Color: GetTrueColorRgb returns 0 for indexed color

mme1950, 2023-10-16:
The GetTrueColorRgb method should return the RBG values of the color represented by this Color object.
This works fine if _color holds the true-color value.
If it is an indexed color the method should return the RGB values from the _indexRGB array.
The values _color=0 (IsByBlock) and _color=256 (IsByLayer) must be handled explicitely.

BTW: _color = 0 means ByBlock but also points to 0th entry of _indexRGB i.e. black.

mme1950, 2023-10-18:
The ByLayer and ByBlock color cannot be resolved in the Color class.
So this is the code we propose:

		public ReadOnlySpan<byte> GetTrueColorRgb()
		{
			if (IsTrueColor)
			{
				return new ReadOnlySpan<byte>(BitConverter.GetBytes(_color), 0, 3);
			}
			if (IsByBlock || IsByLayer)
			{
				return null;
			}

			return _indexRgb[_color].AsSpan();
		}
```

mme1950, 2023-10-18:
This should be extended too:
		public override string ToString()
		{
			if (IsTrueColor)
			{
				var parts = GetTrueColorRgb();
				return $"True Color RGB:{parts[0]},{parts[1]},{parts[2]}";
			}
			if (IsByBlock)
			{
				return "By Block";
			}
			if (IsByLayer)
			{
				return "By Layer";
			}

			var partIndexed = _indexRgb[_color].AsSpan();
			return $"Indexed Color: {Index}, RGB:{partIndexed[0]},{partIndexed[1]},{partIndexed[2]}";
		}

DJGosnell, 2023-10-18:
Been away but I've been keeping my eye on this project.  I think that we need to be careful with the as the `Color` struct only provides either a True or Indexed color.  A color can be either True or Indexed but not both and this suggestion seems to mixing intents a bit.

- `ToString`: When a true color is stored it needs to output just the true color info and when an indexed color is stored, it needs to output just the indexed color info. It should not output the Indexed color's RGB equivalent. Eg. `return $"Indexed Color: {Index}";` I agree about the ByBlock and ByLayer portions, can't believe I didn't add those already.
- `GetTrueColorRgb`: This method needs to only return true colors when it is in-fact a true color; not indexed value's equivalent true colors.  If you need to get equivalent true colors, then the `GetIndexRGB` method should be used.  This method is supposed to be the counterpart to the `TrueColor` property.  This property will currently indicate that the value is not a true color (`-1`).

This being said, a new method could be added to get a true color whether or not it is a true color.  Even an extension method could handle this.

mme1950, 2023-10-19:
> This being said, a new method could be added to get a true color whether or not it is a true color. Even an extension method could handle this.

From the perspective of a user of this library I would like to have a method that returns a RGB value regardless what is is internally.

DomCR, 2023-10-21:
To summarize I think that this issue should take care of 2 things (correct me if I'm mistaken):
- Change `ToString()` to reflect the cases `ByLayer` and `ByBlock`.
- Implement a method to return the values RGB independently if is true color or not, this may be done by changing the current method `GetTrueColorRgb()` or creating a new one like `GetRgb()`.

mme1950, 2023-10-21:
Exactly!

closed 2023-10-26 by DomCR