DomCR/ACadSharp #185: ReadCmColor does not create Color with true-color values

mme1950, 2023-10-17:
The method DwgMergedReader.ReadCmColor does not use the read RGB value.

DomCR, 2023-10-17:
Hi @mme1950 

Colors are not fully implemented for Dwg, this method should be revisited and fixed accordingly.

Opening a branch to refactor the colors and finish the implementation.

mme1950, 2023-10-18:
This works for us, but I have no idea how to handle `colorName` and `bookName`:

		public Color ReadCmColor()
		{
			if (!(_mainReader is DwgStreamReaderAC18))
				return _mainReader.ReadCmColor();

			//To read a color in this version file needs to access to the
			//string data section at the same time.

			//CMC:
			//BS: color index(always 0)
			short colorIndex = ReadBitShort();
			//BL: RGB value
			int rgb = ReadBitLong();

			//RC : Color Byte
			byte id = ReadByte();

			string colorName = string.Empty;
			//RC: Color Byte(&1 => color name follows(TV),
			if ((id & 1) == 1)
				colorName = ReadVariableText();

			string bookName = string.Empty;
			//&2 => book name follows(TV))
			if ((id & 2) == 2)
				bookName = ReadVariableText();

			short i = (short)(rgb & 0xFF);
			if (i >= 0 && i <= 255)
			{
				return new Color(i);
			}

			byte b = (byte)(rgb & 0xFF);
			byte g = (byte)((rgb >> 8) & 0xFF);
			byte r = (byte)((rgb >> 16) & 0xFF);

			return new Color(r, g, b);
		}