DomCR/ACadSharp #197: ReadEnColor creates bad Color

mme1950, 2023-11-06:
We read a complex DWG with several colors. 
It worked "quite well" before. But now it seems all colors are ByBlock.

ReadEnColol has been changed recently:

old: color = new Color((short)(size & 0b111111111111));
new: color = new Color((byte)(size & 0b111111111111));

Rolling back to the old code seems to fix the problem.

BTW: These lines cannot be correct: A true color using the RGB value cannot be created this way:

//Next value is a BS containing the RGB value(last 24 bits).
color = new Color((byte)this.ReadBitLong());

DomCR, 2023-11-06:
Hi @mme1950 

I've been working on the colors in this PR:
* https://github.com/DomCR/ACadSharp/pull/187

There is an investigation of how the color is stored in `ReadCmColor` and how the color is stored, it seems that the documentation don't cover this.

I'll open a branch to check the current implementation and add some Asserts to check the entity color and transparency.

mme1950, 2023-11-11:
Sorry to say: The color is still not OK.

This line should be changed:
color = new Color((**byte**)(size & 0b111111111111));

This seems to fix the problem:
color = new Color((**short**)(size & 0b111111111111));

mme1950, 2023-11-11:
I do not understand the details.
readCmColor has different implementations in DwgMergedReader and in DwgStreamReaderAC18.

I changed these lines in DwgMergedReader:

				//CECOLOR:
				//3221225472
				//0b11000000000000000000000000000000
				//0b1100_0000_0000_0000_0000_0000_0000_0000 --> this should be ByLayer
				//0xC0000000

				//True color
				//color = new Color(arr[0], arr[1], arr[2]);
				uint trueColor = (uint)((int)rgb - 0xC2000000);

				//Needs the check just in case the flag is not set
				if (trueColor < (1 << 24))
				{
					color = Color.FromTrueColor(trueColor);
				}
Now I get the color as before.

DomCR, 2023-11-14:
I need to read more documentation about how the colors are encoded and used in the DWG files, I'll keep this open and see what I can find out.

closed 2023-11-28 by DomCR