DomCR/ACadSharp #206: Read EVALUATIONGRAPH and BLOCKVISIBILITYPARAMER

mme1950, 2023-11-20:
We have implemented readers for EVALUATIONGRAPH and BLOCKVISIBILITYPARAMER.
The implementation is sufficient for our needs, but not complete, and it is hard to make it complete.

see https://github.com/mme1950/ACadSharp/issues/2 for details.

@DomCR woud you suggest to integrate it?

DomCR, 2023-11-20:
Hi @mme1950,

I've been thinking about the feature DWG to SVG and how to implement it but right now my focus is on the release of the Beta for this project.

This would be really great to have in `ACadSharp`, feel free to open a PR and we can discuss the integration to the project and if I can help with anything let me know.

mme1950, 2023-11-20:
Hi @DomCR,

we are about to publish our converter, I hope tomorrow. I will keep you informed.
The solution includes a converter library and a small application to view and edit the SVG and to display it in an intergrated browser.

So we will open a PR for this issue.

mme1950, 2023-11-21:
Hi @DomCR, see https://github.com/nanoLogika.

closed 2025-01-04 by DomCR