DomCR/ACadSharp #222: Handles in Extended Data

mme1950, 2023-12-06:
I created a linear dimension line with alternative arrow symbols.
The handles of the blocks containing the data of the arrow symbols are provided as extended data.
See DXF:

| Group Code | Value | Remark |
| ---- | ---- | ---- |
|1001|ACAD||
|1000|DSTYLE||
|1002|{||
|1070|   343||
|1005|396|Handle of Block _BoxBlank found in this DXF, ACadSharp value: 10809483530619322368 = 0X9603000000000000|
|1070|  173||
|1070| 1||
|1070| 344||
|1005|404|Handle of Block _Dot found in this DXF, ACadSharp value: 289356276058554368 = 0X404000000000000|
1002|}||

In the extended data read by from DWG most data are the same, the handles, however, look different.

mme1950, 2023-12-06:
[20231205_dimension_arrow1_quadrat_arrow2_punkt.zip](https://github.com/DomCR/ACadSharp/files/13582962/20231205_dimension_arrow1_quadrat_arrow2_punkt.zip)

mme1950, 2023-12-06:
Flipping the bytes seems to solve the problem:

96 03 00 00 00 00 00 00 --> 00 00 00 00 00 00 03 96 = 396 
04 04 00 00 00 00 00 00 --> 00 00 00 00 00 00 04 04 = 404

DomCR, 2023-12-08:
I haven't done much work with extended data besides implement the raw reader but seems that this issue is related to the Endian order of the bytes.

Opening a branch to take a look, thanks!

DomCR, 2023-12-08:
> 4 (1004) Binary chunk. The first byte of the value is a char giving the length; the bytes follow.

This is from the OpenDesign document, I'm guessing that this color is stored like this, but I don' know how can we identify that that array is a color or something else, or even if there is more information in it, maybe it contains the transparency too but this will require a manual check.

mme1950, 2023-12-08:
I know that. I try to handle the following cases:
1. I create DimensionLinear with coloured dimension line selecting an indexed colour. In this case an Extended Data entry for ACAD with the name DSTYLE apperars. A field with the value 177 is followed by a field with the colour index as set in AutoCAD.
2. I choose a true colour for the dimension line. In this case an Extended Data entry for ACAD with the name DSTYLE_TC apperars. The field with the value 177 is followed by a field with the 24-byte-array.

closed 2024-01-11 by DomCR