DomCR/ACadSharp #231: can not read  tolerance of Dimension

Labrobot, 2023-12-19:
autocad can read tolerance of Dimension by Dimension.Dimtol，Is there any plan to add this function

DomCR, 2023-12-19:
Hi @Labrobot,

Are you referring to the [Tolerance](https://help.autodesk.com/view/OARX/2021/ENU/?guid=GUID-ADFCED35-B312-4996-B4C1-61C53757B3FD) entity? if that's the case I can take a look to implement it.

Labrobot, 2023-12-19:
[image]
I want read tolerance value of dimension like 200±0.02，200 is value ,±0.02 is the tolerance

Labrobot, 2023-12-19:
[image]
[image]

I also need the function of reading tolerance like picture, if you can implement it that would be better

DomCR, 2023-12-19:
I can implement the tolerance entity, I'll open a branch.

For the image you will need to use some kind of library to process it, the picture that you are showing is created by Autocad and `ACadSharp` does not have tools for image processing.

Labrobot, 2023-12-20:
[image]
Most parts have tolerances of the dimensions and I think it is a defect if the tolerance cannot be read

closed 2024-01-04 by DomCR