DomCR/ACadSharp #241: ATTDEF Value Always Equal to Default

prestsauce, 2024-01-05:
Apologies if this has been asked before. I looked through the issues and the only other issue that seems maybe related is #221, so perhaps this type of block object just isn't compatible with the dxf format. If that's the case, please just let me know and close the issue. However, I'm really just looking for the text values on the autocad block object. Maybe it's not stored on the individual ATTDEF entities?

**Describe the bug**
When reading a DWG file, the Value property of an ATTDEF object always returns the default value for the autocad block attribute. 

**To Reproduce**
It doesn't look like I can upload a sample DWG, but in short, I have an autocad block with attributes. This, for example:
[image]
and I want to get the values that have been set. But when reading the file, I only see the default values. I'd be happy to send over an example file if need be.

**Expected behavior**
A method to read the user-set values on the attributes of the block. 

**Screenshots**
-

**Additional context**
-

DomCR, 2024-01-06:
Hi @prestsauce,

The issues are not related and you don't have to worry about asking, if there is an existing issue that may help you I'll show you the link to it so feel free to ask anything.

About your issue, I think that you are looking in the `Blocks` table instead of the Insert entities, it would help to have a sample of the code that you are using to confirm that but this is how you can find the values in the attributes:

- `CadDocument.Blocks` > this is the table of blocks where you have your "blueprints" to create the `Insert` entities, so all the attributes in the blocks are `AttributeDefinitions` which are used as a template to instantiate the attribute in the model.
- `CadDocument.Entities` > here is where your drawing is, all the objects that you can see in the drawing are here, to find the attributes with the value set you need to look for the entities `Insert`, which are the block references that are drawn in you file, there you only need to navigate throw the `Attributes` list in the Insert and get the values that are stored in the property `Value`.

Here is a quick code example:

```C#
	foreach (var item in doc.Entities.OfType<Insert>())
	{
		foreach (var att in item.Attributes)
		{
			Console.WriteLine(att.Value);
		}
	}
```

Hope this helps, let me know if you need anything else.

prestsauce, 2024-01-08:
Thanks for the quick response. I was viewing the entities, which is why it was so confusing. 

Turns out it's an issue with how AutoCAD saves the file when a single block is saved using the WBLOCK command. The newly created file is the raw block definition (not sure if the terminology here is correct or not). When reading that file, I'm getting the raw values, which I would say is expected behavior in that case. 

I'm going to close this. Thank you for your help!

closed 2024-01-08 by prestsauce