DomCR/ACadSharp #245: Write DWG with ACadSharp , Autocad = This drawing was saved with a student Version

sigster, 2024-01-10:
`    Dim doc As ACadSharp.CadDocument = New ACadSharp.CadDocument()
    Dim writer As ACadSharp.IO.DwgWriter = New ACadSharp.IO.DwgWriter("Test1.dwg", doc)
    writer.Write()
`

I get this messages in Autocad when I create DWG file with ACadSharp

This drawing was saved with a student Version of the product and marked with a plot stamp

DomCR, 2024-01-11:
Hi @sigster,

This is a known issue, the `DwgWriter` is one of the newest features in the project and still needs to be polished.
Autocad seems to use some kind of watermark in the files that is difficult to replicate, I'll open a branch to take a look.

DomCR, 2024-01-11:
If you save the file manually do you still have the plot stamp?

sigster, 2024-01-11:
Do you mean if I save new file with Autocad no I have register Austocad
I don't have student version

DomCR, 2024-01-11:
I mean, if you save the file created with ACadSharp using Autocad, will this re-saved file still have the stamp?

I want to know if it is something that sticks in the file or it can be removed by re-saving the file in a licensed Autocad.

sigster, 2024-01-11:
If I open the file with autocad and save it I still get theis messages
same if I use SAVEAS

[image]

DomCR, 2024-01-11:
Ok, thanks for the info

DomCR, 2024-01-26:
I found a bit of information about this issue:
https://help.autodesk.com/view/ACDLT/2024/ENU/?guid=GUID-A6E37575-FA05-4192-BE84-F19F8C0CE56A
It seems that for version of Autocad later than 2022 the student version won't be relevant anymore, I know that this does not fix the issue but is a workaround by now.

DomCR, 2024-01-26:
I think that was it!

Thanks for the clue!

closed 2024-01-26 by DomCR

DomCR, 2024-01-26:
The change is in master, please if you can confirm that the change fixed the issue.

I'm currently using a student version so it's quite hard for me to confirm that this has been completely fixed.

Thanks!

sigster, 2024-01-26:
This did fix it !

Thanks DomCr

to skip this messages in Autocad then you can use command DWGCHECK = 0

[image]