DomCR/ACadSharp #246: read big dwg file take so long time

brull, 2024-01-11:
when i run this program:
`var sw = new Stopwatch();
sw.Start();
using (DwgReader reader = new DwgReader(_file))
{
	doc = reader.Read();
}
sw.Stop();
Console.WriteLine(sw.ElapsedMilliseconds);`
it takes about 80175ms to read a dwg file about 24M size.

brull, 2024-01-11:
When I use Teigha to read the same file, it takes just 2187ms. The big dwg files are usual on building design. So can you guys improve it. Thanks a lot！

DomCR, 2024-01-11:
Hi @brull,

Can you provide a file as an example? it would be good to detect bottlenecks and have an example.

Thanks.

brull, 2024-01-12:
> This project is still actively being developed so performance is still being worked on. Since this is an open source project and you want use it, you can create a PR and assist with that performance endeavor and everyone will appreciate it. There are some areas where the parsing can improve and I'm working on ways to help with that.

This project is great！ U are great men！Cause it‘s the only one open source project written by C#. Hope it become better and better. 
Now， I am working on rising performance on my job.

brull, 2024-01-12:
> Hi @brull,
> 
> Can you provide a file as an example? it would be good to detect bottlenecks and have an example.
> 
> Thanks.

I'm sorry, the dwg file designed by another guy, I have no idea can I give it to u. But I think this problem is universal. U can find a big dwg file and try it.

closed 2024-06-11 by DomCR