DomCR/ACadSharp #250: Error converting from DWG to DXF

ShadowGreg, 2024-01-23:
I'm trying to convert a DWG to DXF. But the file saved is corrupted.

 public static void Main(string[] args) {
        var path = Ui.Prompt("Enter path to dwg file: ");
        CadDocument doc = DwgReader.Read(path);
        doc.Header.Version = ACadVersion.AC1018;
        var dxfPath = Ui.Prompt("Enter path to dxf file: ");
        using (DxfWriter writer = new DxfWriter(dxfPath, doc,false))
        {
            writer.OnNotification += NotificationHelper.LogConsoleNotification;
            writer.Write();
        }
        //DxfWriter.Write(dxfPath, doc, false);
        
    }

[image]

ShadowGreg, 2024-01-23:
[4550.50.GS00.008.0155.362675.20.133AI.Sh01-37.Rev_a0.dwg.zip](https://github.com/DomCR/ACadSharp/files/14021814/4550.50.GS00.008.0155.362675.20.133AI.Sh01-37.Rev_a0.dwg.zip)

DomCR, 2024-01-23:
Your file seems to have characters in the `Text` and `MText` that the writer cannot process correctly, I'll create another branch to take a look at this second issue.

DomCR, 2024-01-26:
PR that fixed the text format for `Tolerance` and `MText` but the dxf file produced by `ACadSharp` is still invalid in this particular case.

- https://github.com/DomCR/ACadSharp/pull/253