DomCR/ACadSharp #255: Grid off when I create dwg file

sigster, 2024-01-26:
Hi

how can I have Grid off when I create dwg file 

[image]

DomCR, 2024-01-27:
Hi @sigster 

To disable the Grid check the *Active `VPort` and disable the `ShowGrid` flag.

sigster, 2024-01-27:
Hi

Any change you can help me with sample code

Regards
Sigster

DomCR, 2024-01-29:
```C#
	CadDocument doc;
	using (DwgReader reader = new DwgReader(_file))
	{
		doc = reader.Read();

		//This sets the grid off for the *Active VPort
		doc.VPorts[VPort.DefaultName].ShowGrid = false;

		//This guarantees that all VPort configurations have the grid off
		foreach (VPort vp in doc.VPorts)
		{
			vp.ShowGrid = false;
		}
	}
```

Let me know if this helps.

sigster, 2024-01-29:
This works
Thanks ComCR

closed 2024-01-29 by sigster