DomCR/ACadSharp #259: Get image count from "ACAD_IMAGE_DICT"

TheRaikoz, 2024-01-29:
Hi,
I get a dictionary with images, but the count is 0.
There are 6 of them in the document, something is wrong.
ACAD 2017
Please help
[image]
[image]

DomCR, 2024-01-29:
Hi @TheRaikoz,

Do you have a file that I can use as an example to test your case? It would help a lot to spot the issue.

Thanks.

TheRaikoz, 2024-01-29:
Yes, here you go.
[Test.zip](https://github.com/DomCR/ACadSharp/files/14082506/Test.zip)

TheRaikoz, 2024-01-30:
@DomCR
 I added acad Entity ImageDef and everything worked for me

DomCR, 2024-01-31:
I've checked the logs from the reader:

Unlisted object not implemented, DXF name: IMAGE
Unlisted object not implemented, DXF name: IMAGEDEF

I'm guessing that these are the objects that you are looking for, I'll open 2 branches and see if they can be implemented in the reader.

closed 2024-02-05 by TheRaikoz