DomCR/ACadSharp #260: MultiLeaderStyle is not a TableEntry

DomCR, 2024-01-29:
**Describe the bug**
`MultiLeaderStyle` should not inherit the `TableEntry` object.

**Expected behavior**
`MultiLeaderStyle` has to be implemented as an Object nested in a dictionary.

DomCR, 2024-01-29:
@mme1950
@nka1994 

This issue affects part of you implementation, I'll give it some time in case you want to check the changes.

The only major change, besides the namespaces for some classes, is that the MULTILEADERSTYLE is not a table entry, is a dictionary entry.

DomCR, 2024-01-29:
If you check the DXF documentation you will be able to see that the tables in a DXF file are the following:

[APPID (DXF)](https://help.autodesk.com/view/OARX/2021/ENU/?guid=GUID-6E3140E9-E560-4C77-904E-480382F0553E)
[BLOCK_RECORD (DXF)](https://help.autodesk.com/view/OARX/2021/ENU/?guid=GUID-A1FD1934-7EF5-4D35-A4B0-F8AE54A9A20A)
[DIMSTYLE (DXF)](https://help.autodesk.com/view/OARX/2021/ENU/?guid=GUID-F2FAD36F-0CE3-4943-9DAD-A9BCD2AE81DA)
[LAYER (DXF)](https://help.autodesk.com/view/OARX/2021/ENU/?guid=GUID-D94802B0-8BE8-4AC9-8054-17197688AFDB)
[LTYPE (DXF)](https://help.autodesk.com/view/OARX/2021/ENU/?guid=GUID-F57A316C-94A2-416C-8280-191E34B182AC)
[STYLE (DXF)](https://help.autodesk.com/view/OARX/2021/ENU/?guid=GUID-EF68AF7C-13EF-45A1-8175-ED6CE66C8FC9)
[UCS (DXF)](https://help.autodesk.com/view/OARX/2021/ENU/?guid=GUID-1906E8A7-3393-4BF9-BD27-F9AE4352FB8B)
[VIEW (DXF)](https://help.autodesk.com/view/OARX/2021/ENU/?guid=GUID-CF3094AB-ECA9-43C1-8075-7791AC84F97C)
[VPORT (DXF)](https://help.autodesk.com/view/OARX/2021/ENU/?guid=GUID-8CE7CC87-27BD-4490-89DA-C21F516415A9)

For older versions the Viewports are also stored in a table but not anymore.

As you can see there is a table for DimensionStyles but not for MultiLeaderStyles, happens something similar for Layouts, MLineStyles or even VisualStyles, you can check the `CadDictionary` file in this project and check for the root dictionary entries section, there I've added a constant with each entry in the root dictionary.

I don't know why they have decided to implemented this way but that's how the file is organized, so MultiLeaderStyles will be placed in the dictionary ACAD_MLEADERSTYLE.

If you want you can confirm this by finding the actual owner in the `CadMLeaderStyleTemplate`, use the `OwnerHandle` to find the actual owner and you will be able to confirm that is a dictionary.

Sorry that I didn't spot this during the review, but is a change that needs to be added to master for a correct implementation.

Let me know about your thought on this.

mme1950, 2024-01-29:
Oh, we simply did not know. You created a PR already, so we do not have to do anything?

DomCR, 2024-01-29:
I wanted to contact you so you are aware of the changes and just in case that I've haven't missed anything.

So if it's ok for you I'll push the changes with the implementation.

mme1950, 2024-01-30:
> So if it's ok for you I'll push the changes with the implementation.

It is OK.

closed 2024-01-31 by DomCR