DomCR/ACadSharp #268: Create new DWG, Color control and Line Type Control

sigster, 2024-01-31:
When I create new DWG

Color control = Color 0
Shouldn't be = Bylayer as default ?

Line Type Control = notthing is default 
Shouldn't be = Bylayer as default ?

Model and Layout1
Layout1 is Default activate
Shouldn't be = Model as default  ?

Regards
Sigster

DomCR, 2024-02-01:
Hi @sigster 

What do you mean by control?
Are you referring to the header variables?

sigster, 2024-02-01:
This , when I open DWG file

[image]

DJGosnell, 2024-02-01:
I believe you are mixing up concerns. This project gives you exact control over the DWG/DXF file reading and writing so you can make the layers or entities do whatever you want. If you are looking for GUI controls, that is out of scope of this project.  I'd suggest learning about how DWG/DXF files are used with the documentation in the Docs directory.

DomCR, 2024-02-01:
These values are set by default but they are not constraint to anything in particular.

If you want a sort of defaults for your document you can change the values in the header of the document, here is an example with the `LineType` by default:

```C#
doc.Header.CurrentLineTypeName = LineType.ByBlockName;
```

Not sure which variable holds the current Layout, I would have to check that.

sigster, 2024-02-01:
Thanks

Do you have sample how to change Header color to Bylayer

[image]

DomCR, 2024-02-01:
It seems that, there is a limitation with colors, the colors in the header are written in a different way than the other entities and by now the writer cannot write the colors in the header property, so by now I think that this value cannot change.

I'll open a branch assigned to this issue to take a look at this bug.

sigster, 2024-02-01:
Thanks

and thanks for the

`doc.Header.CurrentLineTypeName = LineType.ByBlockName;`

Regards
Sigster

closed 2024-08-15 by DomCR

sungaoyong, 2024-12-17:
> Model and Layout1
> Layout1 is Default activate
> Should it be = Model as default ?

 document.Header.ShowModelSpace = true;