DomCR/ACadSharp #274: BlockEntity.BasePoint not assigned correctly while reading DWG

sv-ed, 2024-02-07:
**Describe the bug**
I'm trying to obtain the insert point of a block that contains some etntites but I cannot find an easy way to do it.
I tried reading the blocks BasePoint but seems not to be set while reading the file and all blocks have the BasePoint set to the default value (XYZ.Zero).
The Insert Entities seems to read correctly their InsertPoint but I cannot find a way to link the entity contained in the block to the insert

**To Reproduce**
Load a DWG file (mine is AC1032 btw) and check the Blocks contained in the BlockRecord Table

**Expected behavior**
The BasePoint of the Block entity should be different than the default value (XYZ.Zero)

**Screenshots**
[image]

DomCR, 2024-02-07:
Hi @sv-ed 

A `BlockRecord` is a reference of a block in the document, not it's instance, you need to iterate in the drawing entities and retrieve the `Insert` entities.

If you are looking for a specific block, you can use this code:

```C#
		/// <summary>
		/// Get all the instances of an specific block in the model
		/// </summary>
		/// <param name="file"></param>
		/// <param name="blockname"></param>
		/// <returns></returns>
		public static IEnumerable<Insert> GetInsertEntities(string file, string blockname)
		{
			CadDocument doc = DwgReader.Read(file);

			// Get the insert instance that is using the block that you are looking for
			return doc.Entities.OfType<Insert>().Where(e => e.Block.Name == blockname);
		}
```

Once you have the `Insert` entities you can look for the `InsertPoint` property, which is the point that locates the insert in the model.

Let me know if that helped you.

sv-ed, 2024-02-07:
Hi @DomCR, thanks for the response.
I've made a slight modification to the code to make it works with the `Handle` instead of the `Name` to be more specific and it works fine.
Thank you!

closed 2024-02-07 by DomCR