DomCR/ACadSharp #277: Mtext ruler and the mouse cursor is vertical

sigster, 2024-02-08:
Hi

When I create Mtext I get The ruler and the mouse cursor is vertical 

Are others experiencing this or is it just me?

---

` 
  doc.Header.Version = ACadVersion.AC1018
  doc.Header.PointDisplaySize = 5.0
  doc.Header.PointDisplayMode = 35
  doc.Header.CurrentLineTypeName = LineType.ByLayerName

  doc.VPorts(VPort.DefaultName).ShowGrid = False

  For Each vp As VPort In doc.VPorts
      vp.ShowGrid = False
  Next

  Dim mtext As MText = New MText()
  mtext.Value = "Test"
  doc.Entities.Add(mtext)

  Using writer As DwgWriter = New DwgWriter(dwgFilePath22, doc)
      writer.Write()
  End Using`

[image]

sigster, 2024-02-08:
When I drag the corner to resize 
Drag the mouse right the Hight number change
Drag the mouse down the width number change

it is like the frame has twist to right

[image]

DomCR, 2024-02-09:
Hi @sigster 

I've tried to recreate this issue on my end, but I don't get the same result, maybe is for the editor that you are using.

There are a couple of things that you can try:
- Your insert point seems to be set at `BottomLeft`, but the default is `TopLeft`, check the `AttachmentPoint` property for `MText` and try to change it and see if it changes the drawing.
- There is also the property `DrawingDirection` for me it takes the value `ByStyle`, if that's the case check the TextStyle "Standard" in the Table, there are a couple of properties there that may solve your issue like the flags or the mirror settings.

Hope this helps.

sigster, 2024-02-09:
If I use the command in Autocad AUDIT
it fix the Mtext

Auditing Header
Auditing Tables
Auditing Entities Pass 1

AcDbLayerTableRecord: "0"   Color Out of Range          7
AcDbMText(1C)                     was repaired.
Pass 1 100     objects audited
Auditing Entities Pass 2

Pass 2 100     objects auditedAcDbViewport(76) Invalid linetype eNullObjectId           changed to ByLayer
Auditing Blocks
 0       Blocks audited
Auditing AcDsRecords
Total errors found 4 fixed 4

sigster, 2024-02-09:
Thanks 

it look like this tix it

mtext.DrawingDirection = DrawingDirectionType.LeftToRight

closed 2024-02-09 by sigster